Staubgeborener/Klipper-Backup #123: Installing on new board causes old backups to be deleted.

stevemalsam, 2024-09-04:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem and that I have tested the `--fix` command mentioned above without success.

### What happened

I had Klipper-Backup running on a BTT CB1 on a BTT Manta m8p. I upgraded the CB1 to a Raspberry Pi. In the process of installing things, I installed Klipper-Backup, and pointed it to the same repository. I suddenly get errors saying Klipper can't find printer.cfg. 

Turns out, instead of restoring the files from my backup, Klipper-Backup overwrote them, and deleted almost all of my config files. 

### What did you expect to happen

I expected Klipper-Backup to restore my config files from the git repository. 

### How to reproduce

Install Klipper-Backup. Create a repository and back up. On another SD card or another board with a fresh install of Klipper, install Klipper-Backup. 

### Debug

Not relevant.

Tylerjet, 2024-09-04:
There is no restore functionality on main branch, running the script on another device would overwrite the git repo and currently that is expected.

stevemalsam, 2024-09-05:
That really does not make any sense. I cannot see any situation in which I would want my backups to be clobbered like that.

Tylerjet, 2024-09-05:
Restore functionality is in progress, but i'm not sure what lead you to believe just running script.sh again on a different system would restore settings?

stevemalsam, 2024-09-05:
For one, nothing anywhere says there's no restore functionality, and I really can't understand what the point of a backup utility without having a restore functionality is. For two, it's git. You kinda have to go out of your way to check out a repository and the branch, and clobber it.

Tylerjet, 2024-09-05:
With Klipper you can't just clone the repo back, you have to keep track of where the files were pulled from and where they need to be pushed back to, along with if your running a mainsail theme, you have to back that up properly, and same goes for moonraker db. the original idea behind the script was just to backup the files to a git repo and then you the user could download a .zip of the repo and manually copy the files back to restore.

But again i'm not sure where in the documentation provided does it state that running script.sh with no parameters restores the backup.

As for "it's git", I'm not sure what that is supposed to mean, yeah it's git but we perform specific action in order to insure all changes are tracked to the repo, that includes any files that are gone when a new backup is taken.

Tylerjet, 2024-09-05:
You can still go back to the prior commit and download the files from that commit state on github to restore config files on the new device and then run the backup again to push those back to the repo.

stevemalsam, 2024-09-05:
Nowhere in the documentation does it say that there is no restore functionality. And if I am installing it for the first time, then the script is going to be run automatically. And it should be able to tell that there is something in that repository. 

At the very least, the script, when checking out the repository and realizing that it is not empty, should issue a huge warning, saying that the files in this repository will be overwritten, and give the user a chance to cancel. Automatically deleting several files with no warning is not a way to go.

Staubgeborener, 2024-09-05:
Your files aren't deleted, git is using versioning and thus you can simply get your files from your commit history.

closed 2024-10-03 by github-actions[bot]