Staubgeborener/Klipper-Backup #124: fatal: not a git repository (or any of the parent directories): .git

B1ueOne, 2024-09-11:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem and that I have tested the `--fix` command mentioned above without success.

### What happened

Klipper-Backup stopped syncing 5 months ago. Today I decided to try and fix it.

What I've tried:
- updating Klipper-Backup
- running the install script again
- deleting all files and directories that have anything to do with klipper-backup

and yet I keep getting the same error when executing the script:

```
● Checking for installed dependencies Done!

Klipper-Backup is up to date

fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Branch: main in .env does not match the currently checked out branch of: .
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not in a git directory
fatal: not in a git directory
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: No remote configured to list refs from.
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
```

### What did you expect to happen

successful backup to my repo.

### How to reproduce

run the script.sh

### Debug

```
● Checking for installed dependencies Done!

Klipper-Backup is up to date

------------DEBUG:------------
github_token=****************
github_username=B1ueOne
github_repository=KlipperBackup
branch_name="main"
commit_username="pi"
commit_email="pi@VzBot-b4ff468"

# All information regarding .env can be found here:
# https://klipperbackup.xyz/configuration/

# Backup paths
# Note: script.sh starts its search in $HOME which is /home/{username}/
# The array accepts folders or files like the following example
#
#  backupPaths=( \
#  "printer_data/config/*" \
#  "printer_data/config/printer.cfg" \
#  )
#
# Using the above example the script will search for `/home/{username}/printer_data/config/*` and `/home/{username}/printer_data/config/printer.cfg`
# When backing up a folder you should always have `/*` at the end of the path so that files insde the folder are properly searched

backupPaths=( \
"printer_data/config/*" \
)

# Array of strings in .gitignore pattern git format https://git-scm.com/docs/gitignore#_pattern_format for files that should not be uploaded to the remote repo
# New additions must be enclosed in double quotes and should follow the pattern format as noted in the above link
exclude=( \
"*.swp" \
"*.tmp" \
"printer-[0-9]*_[0-9]*.cfg" \
"*.bak" \
"*.bkp" \
"*.csv" \
"*.zip" \
------------------------------

------------DEBUG:------------
$HOME: /home/pi
------------------------------

------------DEBUG:------------
$backup_path: /home/pi/config_backup

Content of $backup_path:
total 16
drwxr-xr-x  3 pi pi 4096 Sep 11 10:25 .
drwxr-xr-x 22 pi pi 4096 Sep 11 09:55 ..
drwxr-xr-x  6 pi pi 4096 Sep 11 07:49 .git
-rw-r--r--  1 pi pi  181 Sep 11 07:49 README.md
------------------------------

------------DEBUG:------------
$backup_path/.git/config:

[init]
    defaultBranch =
------------------------------

fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Branch: main in .env does not match the currently checked out branch of: .
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not in a git directory
fatal: not in a git directory
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg

------------DEBUG:------------
Content of $backup_path after rsync:
total 20
drwxr-xr-x  4 pi pi 4096 Sep 11 10:33 .
drwxr-xr-x 22 pi pi 4096 Sep 11 09:55 ..
drwxr-xr-x  6 pi pi 4096 Sep 11 07:49 .git
drwxr-xr-x  3 pi pi 4096 Sep 11 10:33 printer_data
-rw-r--r--  1 pi pi  181 Sep 11 07:49 README.md
------------------------------

fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: No remote configured to list refs from.
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git

```

Tylerjet, 2024-09-11:
> ```
> ------------DEBUG:------------
> $backup_path/.git/config:
> 
> [init]
>     defaultBranch =
> ------------------------------
> ```

In that file defaultBranch should not be blank and should be matching the same name set in .env. 

```

> # Array of strings in .gitignore pattern git format https://git-scm.com/docs/gitignore#_pattern_format for files that should not be uploaded to the remote repo
> # New additions must be enclosed in double quotes and should follow the pattern format as noted in the above link
> exclude=( \
> "*.swp" \
> "*.tmp" \
> "printer-[0-9]*_[0-9]*.cfg" \
> "*.bak" \
> "*.bkp" \
> "*.csv" \
> "*.zip" \

```
This should have an ending parenthesis is it also missing from the .env file?

Staubgeborener, 2024-09-11:
This is a weird thing:
```
------------DEBUG:------------
$backup_path/.git/config:

[init]
    defaultBranch =
------------------------------
```

I just deleted my whole klipper-backup and config_backup folder, cloned this repository and run the `install.sh`. After running `script.sh` my .git config looks like this:
```
------------DEBUG:------------
$backup_path/.git/config:

[init]
    defaultBranch = main
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[user]
        name = klipperbackup
        email = klipperbackup@kubuntu-vm
[remote "origin"]
        url = https://********@github.com/Staubgeborener/klipperbackupstagingtest.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[branch "main"]
        remote = origin
        merge = refs/heads/main
------------------------------
```

And for context, running `script.sh` works like a charm:
```
● Checking for installed dependencies Done!

Klipper-Backup is up to date

Von https://github.com/Staubgeborener/klipperbackupstagingtest
 * branch            main       -> FETCH_HEAD
Bereits aktuell.
Auf Branch main
Ihr Branch ist auf demselben Stand wie 'origin/main'.

nichts zu committen, Arbeitsverzeichnis unverändert
[main 767a1d9] New backup from 11.09.2024 - 12:32:02 - No new changes pushed
Objekte aufzählen: 1, fertig.
Zähle Objekte: 100% (1/1), fertig.
Schreibe Objekte: 100% (1/1), 217 Bytes | 217.00 KiB/s, fertig.
Gesamt 1 (Delta 0), Wiederverwendet 0 (Delta 0), Pack wiederverwendet 0
To https://github.com/Staubgeborener/klipperbackupstagingtest.git
   e65ccfd..767a1d9  main -> main
Branch 'main' folgt nun 'origin/main'.
```

The `script.sh` just checks if the `.git` directory exists and initialize it otherwise ([source](https://github.com/Staubgeborener/Klipper-Backup/blob/999e8143657ad2ef63312532f6c027d954ddd985/script.sh#L126)). My first idea, there is a old and wrong `.git/config` file in your `config_backup` folder. Did you really run `sudo rm -rf ~/config_backup` or maybe something like
```
klipperbackup@kubuntu-vm:~$ cd ~/backup_config
klipperbackup@kubuntu-vm:~/config_backup$ rm *
```
as this will not delete hidden files and directory like `.git`?

Otherwise run all of these commands in the same order and show me the whole output:
```
cat ~/config_backup/.git/config
~/klipper-backup/script.sh --fix
~/klipper-backup/script.sh
```

@Tylerjet I don't think the ending parenthesis is truly missing, i also don't get it with the `--debug` output, as there seems to be a small issue in implementing this parameter.  Need to fix this.

B1ueOne, 2024-09-11:
```
pi@VzBot:~ $ cat ~/config_backup/.git/config
[init]
    defaultBranch =
pi@VzBot:~ $ ~/klipper-backup/script.sh --fix
● Checking for installed dependencies Done!

● Delete config_backup folder Done!

pi@VzBot:~ $ ~/klipper-backup/script.sh
● Checking for installed dependencies Done!

Klipper-Backup is up to date

Initialized empty Git repository in /home/pi/config_backup/.git/
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
remote: Enumerating objects: 616, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (29/29), done.
remote: Total 616 (delta 18), reused 35 (delta 10), pack-reused 570 (from 1)
Receiving objects: 100% (616/616), 187.14 KiB | 2.40 MiB/s, done.
Resolving deltas: 100% (372/372), done.
From https://github.com/B1ueOne/KlipperBackup
 * branch            main       -> FETCH_HEAD
 * [new branch]      main       -> origin/main
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg
[main 064263e] New backup from 11/09/24 - 12:49:52
 6 files changed, 676 insertions(+), 591 deletions(-)
 create mode 100644 printer_data/config/crowsnest.conf
 create mode 100644 printer_data/config/moonraker.conf.backup
 rewrite printer_data/config/printer.cfg (72%)
 delete mode 100644 printer_data/config/spyglass.conf
From https://[MYTOKEN]@github.com/B1ueOne/KlipperBackup.git
Enumerating objects: 15, done.
Counting objects: 100% (15/15), done.
Delta compression using up to 4 threads
Compressing objects: 100% (8/8), done.
Writing objects: 100% (9/9), 15.35 KiB | 2.19 MiB/s, done.
Total 9 (delta 3), reused 0 (delta 0), pack-reused 0
remote: Resolving deltas: 100% (3/3), completed with 3 local objects.
To https://github.com/B1ueOne/KlipperBackup.git
   1f3cf80..064263e  main -> main
Branch 'main' set up to track remote branch 'main' from 'origin'.
pi@VzBot:~ $

```

Strange, now it works somehow?

Staubgeborener, 2024-09-11:
Well, the `--fix` parameter deleted your whole `config_backup` folder with a simple `rm -rf $HOME/$backup_folder` command. After that, running `script.sh` leads the code into creating a fresh new `.git/config` file as the old one is deleted.
I can only assume, that you didn't really deleted it with `rm -rf ~/config_backup` but with a `cd ~/config_backup && rm *`'ish command, which means you didn't deleted the hidden content inside this directory.

But glad its working on your side.

closed 2024-09-11 by Staubgeborener