Staubgeborener/Klipper-Backup #125: updated to v1.3.1-11-gf7bb5204 and now when I press macro I get error

Bradford1040, 2024-09-14:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem and that I have tested the `--fix` command mentioned above without success.

### What happened

hint: Updates were rejected because the tip of your current branch is behind
hint: its remote counterpart. If you want to integrate the remote changes,
hint: use 'git pull' before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.

### What did you expect to happen

well it to back up like it normally did

### How to reproduce

nothing changed 

### Debug

I am not able to really debug 

### Additional information

any help would be great

Staubgeborener, 2024-09-14:
> I am not able to really debug

Like it's written down there: run `script.sh --debug` and post the output.

I guess you changed something in your github repository, like uploaded a file or changed something. Something like that.
Now your local code doesn't match with the remote one anymore.

Normally the script handles stuff like that. I guess you changed something on your remote repository and also played around with different repositories and/or branches and edited the .env a couple of times while editing the content? Please tell exactly what you did.

Bradford1040, 2024-09-14:
Of course it was my fault, thank you for responding so quickly, it had nothing to do with you I believe, I am pretty sure it had to me renaming the default branch, as I switched to Danger Klipper from main line Klipper and wanted to create a second repo for it and keep the first! But had forgotten that I got a warning about switching the default branch, so pretty sure this can be closed

Bradford1040, 2024-09-14:
also, you said "Like it's written down there: run script.sh --debug and post the output." where did I see that and pass over it? just future reference, and learning stuff

Staubgeborener, 2024-09-14:
> I am pretty sure it had to me renaming the default branch, as I switched to Danger Klipper from main line Klipper and wanted to create a second repo for it and keep the first! But had forgotten that I got a warning about switching the default branch, so pretty sure this can be closed  

I don't think I truly understand what you did, but to break it down: if you rename or create a new branch which you want to use you have to put the name of this new branch also in your `.env` file. If you switch the repository you have to put the name of this new repository also in your `.env`. I guess you mixed something up?

I always ask for as many details as possible because we naturally want to improve the code. There are certain dependencies under which problems arise that we can't even think about. Problems like yours can help to recognize this and add new checks to the code so that we can either fix it directly while the user is running the code or point out a problem to the user with a specific error message.

> also, you said "Like it's written down there: run script.sh --debug and post the output." where did I see that and pass over it? just future reference, and learning stuff

Opening a new issue will explicitly ask for this:

[image]

closed 2024-09-14 by Staubgeborener