Staubgeborener/Klipper-Backup #126: Git Backup corrupted on every backup (triggered by boot.)

jwhite, 2024-09-15:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem and that I have tested the `--fix` command mentioned above without success.

### What happened

First backup triggered normally from the shell works.  

Subsequent backups fail with this error :

error: object file .git/objects/16/f3809f603034562731dc0eb244ad696acbf125 is empty
fatal: loose object 16f3809f603034562731dc0eb244ad696acbf125 (stored in .git/objects/16/f3809f603034562731dc0eb244ad696acbf125) is corrupt

Subsequent backups don't occur with no warning to the user.

This is endlessly repeatable with a long history of occurring.

~/klipper-backup/script.sh --fix
Works temporarily, as it deletes the backup folder.

### What did you expect to happen

Backups to occur.

### How to reproduce

Installed with all backup options available, but this appears to be using the backup on boot when it fails.

Have uninstalled and reinstalled repeatedly.

### Debug

● Checking for installed dependencies Done!

Klipper-Backup is up to date

------------DEBUG:------------
Command: cd KAMP
------------------------------

------------DEBUG:------------
github_token=****************
github_username=jwhite
github_repository=Voron-klipper_config
branch_name="main"
commit_username="pi"
commit_email="pi@voron-12800de"

# Backup paths
#  Note: script.sh starts its search in $HOME which is /home/{username}/
# The array accepts folders or files like the following example
#
#  backupPaths=( \
#  "printer_data/config/*" \
#  "printer_data/config/printer.cfg" \
#  )
#
# Using the above example the script will search for /home/{username}/printer_data/config/* and /home/{username}/printer_data/config/printer.cfg
# When backing up a folder you should always have /* at the end of the path so that files insde the folder are properly searched

backupPaths=( \
 "printer_data/config/*" \
)

# Array of strings in .gitignore pattern git format https://git-scm.com/docs/gitignore#_pattern_format for files that should not be uploaded to the remote repo
# New additions must be enclosed in double quotes and should follow the pattern format as noted in the above link

exclude=( \
 "*.swp" \
 "*.tmp" \
 "printer-[0-9]*_[0-9]*.cfg" \
 "*.bak" \
 "*.bkp" \
 "*.csv" \
 "*.zip" \
)
------------------------------

------------DEBUG:------------
curl: (22) The requested URL returned error: 404
Error: no GitHub repo jwhite/Voron-klipper_config found (maybe private)
------------------------------

------------DEBUG:------------
$HOME: /home/pi
------------------------------

------------DEBUG:------------
$backup_path: /home/pi/config_backup

Content of $backup_path:
total 16
drwxr-xr-x  3 pi pi 4096 Sep 16 09:03 .
drwx--x--x 34 pi pi 4096 Sep 14 16:34 ..
drwxr-xr-x  8 pi pi 4096 Sep 16 09:03 .git
-rw-r--r--  1 pi pi  181 Sep 14 16:21 README.md
------------------------------

------------DEBUG:------------
$backup_path/.git/config:

[init]
    defaultBranch = main
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[user]
        name = pi
        email = pi@voron-12800de
[remote "origin"]
        url = https://********@github.com/jwhite/Voron-klipper_config.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[branch "main"]
        remote = origin
        merge = refs/heads/main
------------------------------

error: object file .git/objects/16/f3809f603034562731dc0eb244ad696acbf125 is empty
fatal: loose object 16f3809f603034562731dc0eb244ad696acbf125 (stored in .git/objects/16/f3809f603034562731dc0eb244ad696acbf125) is corrupt
Skipping symbolic link: printer_data/config/fluidd.cfg
Skipping symbolic link: printer_data/config/KAMP
skipping non-regular file "printer_data/config/mmu/base/mmu_cut_tip.cfg"
skipping non-regular file "printer_data/config/mmu/base/mmu_form_tip.cfg"
skipping non-regular file "printer_data/config/mmu/base/mmu_leds.cfg"
skipping non-regular file "printer_data/config/mmu/base/mmu_sequence.cfg"
skipping non-regular file "printer_data/config/mmu/base/mmu_software.cfg"
skipping non-regular file "printer_data/config/mmu/base/mmu_state.cfg"
skipping non-regular file "printer_data/config/mmu/optional/client_macros.cfg"
skipping non-regular file "printer_data/config/mmu/optional/mmu_ercf_compat.cfg"
skipping non-regular file "printer_data/config/mmu/optional/mmu_menu.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/base/mmu_cut_tip.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/base/mmu_form_tip.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/base/mmu_leds.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/base/mmu_sequence.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/base/mmu_software.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/base/mmu_state.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/optional/client_macros.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/optional/mmu_ercf_compat.cfg"
skipping non-regular file "printer_data/config/mmu-20240826_102122/optional/mmu_menu.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/base/mmu_cut_tip.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/base/mmu_form_tip.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/base/mmu_leds.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/base/mmu_sequence.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/base/mmu_software.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/base/mmu_state.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/optional/client_macros.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/optional/mmu_ercf_compat.cfg"
skipping non-regular file "printer_data/config/mmu-20240902_174632/optional/mmu_menu.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/base/mmu_cut_tip.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/base/mmu_form_tip.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/base/mmu_leds.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/base/mmu_sequence.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/base/mmu_software.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/base/mmu_state.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/optional/client_macros.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/optional/mmu_ercf_compat.cfg"
skipping non-regular file "printer_data/config/mmu-20240906_113652/optional/mmu_menu.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/base/mmu_cut_tip.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/base/mmu_form_tip.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/base/mmu_leds.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/base/mmu_sequence.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/base/mmu_software.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/base/mmu_state.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/optional/client_macros.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/optional/mmu_ercf_compat.cfg"
skipping non-regular file "printer_data/config/mmu-20240914_155223/optional/mmu_menu.cfg"

------------DEBUG:------------
Content of $backup_path after rsync:
total 20
drwxr-xr-x  4 pi pi 4096 Sep 16 09:05 .
drwx--x--x 34 pi pi 4096 Sep 14 16:34 ..
drwxr-xr-x  8 pi pi 4096 Sep 16 09:05 .git
drwxr-xr-x  3 pi pi 4096 Sep 16 09:05 printer_data
-rw-r--r--  1 pi pi  181 Sep 14 16:21 README.md
------------------------------

error: object file .git/objects/16/f3809f603034562731dc0eb244ad696acbf125 is empty
error: object file .git/objects/16/f3809f603034562731dc0eb244ad696acbf125 is empty
fatal: loose object 16f3809f603034562731dc0eb244ad696acbf125 (stored in .git/objects/16/f3809f603034562731dc0eb244ad696acbf125) is corrupt
error: object file .git/objects/16/f3809f603034562731dc0eb244ad696acbf125 is empty
error: object file .git/objects/16/f3809f603034562731dc0eb244ad696acbf125 is empty
fatal: loose object 16f3809f603034562731dc0eb244ad696acbf125 (stored in .git/objects/16/f3809f603034562731dc0eb244ad696acbf125) is corrupt
branch 'main' set up to track 'origin/main'.
Everything up-to-date

Tylerjet, 2024-09-16:
This issue is likely due to your MMU files that are attempting to be backed up but are symbolic links, when you look at your backup repo is there a folder that looks to be hyperlinked to a location that doesn't exist?
[image]

When the script tries to pull the repo back down it is probably viewing those locations as empty/corrupt as they have no data.

Could you trt what Staubgeborener suggests here: https://github.com/Staubgeborener/Klipper-Backup/issues/121#issuecomment-2345459135 and see if you get the same errors with the files ignored.

I would make sure to use a new empty repo as well for the test.

Staubgeborener, 2024-09-16:
This is one of the best examples why I add this "I promise that i use markdown syntax" checkbox. Please don't just check this box and that's it - use `markdown` to make this stuff readable.

jwhite, 2024-09-17:
> This is one of the best examples why I add this "I promise that i use markdown syntax" checkbox. Please don't just check this box and that's it - use `markdown` to make this stuff readable.

You should specify "Use markdown when including logs as the introduce special characters."

Tylerjet, 2024-09-17:
Just found out that there is actually an attribute that can be set to just render it into a codeblock automagically so will make that change later.

Staubgeborener, 2024-09-17:
> > This is one of the best examples why I add this "I promise that i use markdown syntax" checkbox. Please don't just check this box and that's it - use `markdown` to make this stuff readable.
> 
> You should specify "Use markdown when including logs as the introduce special characters."

No, i would love to see markdown backticks \`\` everywhere which is code related, including snippets, as it increases readability for everything - not just for log outputs due to special characters. That's just the cherry on top.

closed 2024-10-16 by github-actions[bot]