Staubgeborener/Klipper-Backup #131: it's not backing up

drumstick77, 2024-09-25:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the code more readable which helps to fix the problem and that I have tested the --fix command mentioned above without success.

### What happened

I've reinstalled and followed install.sh, entered my details but it's not backing up. it's now doing it on both printers. 

### What did you expect to happen

I expected it to backup to my github

### How to reproduce

 I've re-installed but it is still not uploading files to my repositories

### Debug

```shell
Unable to create/upload debug
```

### Additional information

[image]

Staubgeborener, 2024-09-25:
> Unable to create/upload debug

 Why?

I think you are using your github password in your .env file instead a [token](https://klipperbackup.xyz/installation/#create-github-token). The error message tells you basically [this](https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls). 
> Password-based authentication for Git has been removed in favor of more secure authentication methods. For more information, see "[Managing your personal access tokens]:

drumstick77, 2024-09-25:
okay, so how would one resolve this issue?

Staubgeborener, 2024-09-25:
You create a token like it is explained [here](https://klipperbackup.xyz/installation/#create-github-token) and use it instead your github password in your .env. You could simple run the `install.sh` and add the token if the installer ask for it.

drumstick77, 2024-09-25:
ok, yes that's exactly how I did it
I created the token in github and pasted it during the wizard of running install.sh

drumstick77, 2024-09-25:
I never had to edit or tweak .env

drumstick77, 2024-09-25:
should I remove the current .env and start again?

Staubgeborener, 2024-09-25:
You can just edit your post instead of re-posting.

> should I be generating tokens (classic)? or fine-grained tokens? because your instruction says doesn't matter which one. and I'm using fine-grained tokens

Indeed, it doesn't matter but just create a Fine-grained token with these settings (i will update the docs with this information):
```
Only select repositories -> select your backup repository

Account permissions:
-------------------------------
Administration: Read and write
Commit statuses: Read and write 
```

> should I remove the current .env and start again?
This is not necessary. Create the token like i explained, run the `install.sh` again (like `~/klipper-backup/install.sh`) and add the token when the script ask for it.

If this still doesn't work, please give me the whole `script.sh --debug` output.

drumstick77, 2024-09-25:
ok, I've managed to fix this. I went with the classic token which gives the ability to set permission like per modbot video. the fine grain is slightly different. Although, if I remember rightly Minimal 3DP went with fine grain and he clicked on generate token without setting any permissions. Anyway, I will go with this classic token and will also apply to my other printers. thanks for you help.

closed 2024-09-25 by Staubgeborener

drumstick77, 2024-09-25:
just did it do another printer and was getting this error 
[image]

Staubgeborener, 2024-09-25:
Try `script.sh --fix` and run again.

drumstick77, 2024-09-25:
Good man! awesome. thank you. all sorted now. Cheers