Staubgeborener/Klipper-Backup #134: allow  ssh:// as part of the URL

vajonam, 2024-11-19:
### Is your feature request related to a problem? Please describe.

I have selfhosted gitea running on an on standard port "222", which means the remote URL needs to look like.

`ssh://git@hal.domain.com:222/gituser/klipper_trident.git`

### Describe the solution you'd like

```
-if [[ $git_protocol == "ssh" ]]; then
-    full_git_url="git@"$git_host":"$github_username"/"$github_repository".git"
-else
-    full_git_url=$git_protocol"://"$github_token"@"$git_host"/"$github_username"/"$github_repository".git"
-fi
+full_git_url=$git_protocol"://"$github_token"@"$git_host"/"$github_username"/"$github_repository".git"
 exclude=${exclude:-"*.swp" "*.tmp" "printer-[0-9]*_[0-9]*.cfg" "*.bak" "*.bkp" "*.csv" "*.zip"}
```

### Describe alternatives you've considered

None

### Additional information

If we move to just url with protocol in the front like `ssh://` it allows me to specify port in the git_host: hal.domain.com:222 which then works well.

Staubgeborener, 2024-11-19:
I don't have a gitea instance to test this locally, but couldn't you declare the `github_token` variable as `git` for a dirty workaround if we just want to achieve `ssh://git@hal.domain.com:222/gituser/klipper_trident.git`? Can you push without a token?

Something like
```shell
github_token=git
github_username=gituser
github_repository=klipper_trident
git_host="hal.domain.com:222"
git_protocol=ssh
```

I cannot currently test whether the normal push via ssh to GitHub also works with an `ssh://`  in front of it, so we can replace `git@"$git_host":"$github_username"/"$github_repository".git"` with `"git_protocol"://git@"$git_host":"$github_username"/"$github_repository".git"`.

If so, I would simply update the line soon, as it shouldn't make any difference for existing users.

vajonam, 2024-11-19:
The issue is that I need the `ssh://`  as well separate the username and the host:port with a slash when specifying a non 22 port. 

if the protocol is ssh the URL is rewritten with very little control.

```
if [[ $git_protocol == "ssh" ]]; then
    full_git_url="git@"$git_host":"$github_username"/"$github_repository".git"
else
    full_git_url=$git_protocol"://"$github_token"@"$git_host"/"$github_username"/"$github_repository".git"
fi
exclude=${exclude:-"*.swp" "*.tmp" "printer-[0-9]*_[0-9]*.cfg" "*.bak" "*.bkp" "*.csv" "*.zip"}

```

so what I am suggesting is the using the full ssh URL instead of the scp like URL syntax.

otherwise I have manage this in the .ssh/config file, which is a possibility but the above seems like much cleaner solution.

vajonam, 2024-11-19:
Also this is not a gitea issue, just a non standard ssh port issue.

vajonam, 2024-11-19:
Think we are on the same page, our messages crossed  :+1:

Staubgeborener, 2024-11-20:
Could you update to the latest version and test it? You should now enter the following in the .env:

```
git_host="hal.domain.com:222"
```

We now use another new variable called `ssh_user`. This is set to `git` by default, as this is the default git user in gitea and always works with GitHub. In your specific case, you should therefore not need this variable at all. However, if you have assigned an individual ssh user to gitea (or whichever service you use), you could now set `ssh_user=“whatever”` inside the .env file.

vajonam, 2024-11-20:
LGTM, works fine. tried with the latest.

vajonam, 2024-11-20:
fixed with 1ccd26a

closed 2024-11-20 by vajonam