Staubgeborener/Klipper-Backup #138: The script doesn't upload any of the conifg files

Voron-Switchwire, 2024-12-12:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the code more readable which helps to fix the problem and that I have tested the --fix command mentioned above without success.

### What happened

I install klipper-backup according to the documentation. At first, it worked, but after I had some difficulties trying to get the autoupdate to work, the script just deleted all the files from my GitHub repository. Since then, if tried to change the .env, uninstalled klipper-backup with kiauh and reinstall it with kiauh. With no success. I don't get, why klipper-backup doesn't see my config files?
I would appreciate any help!

### What did you expect to happen

To get klipper-backup to recognize my files

### How to reproduce

IDK

### Debug

```shell
IDK
```

Staubgeborener, 2024-12-13:
IDK why you wrote IDK everywhere. This makes it difficult to help.
First: Your old backup seems to be [here](https://github.com/Voron-Switchwire/Switchwire/commit/12621bcc08b3fe54dd372f4646697d8d52c5f82e).

Second: Please run `~/klipper-backup/script.sh --debug` as explained in the issue template and post the output (with `markdown` syntax).

Third: Did you really run `~/klipper-backup/script.sh --fix`? Show the output.

Voron-Switchwire, 2024-12-13:
I had some issues with git. The there was a .git file in the config folder. Therefore, the script couldn't initialize the repository. After deleting it manually, it works like a charm. Thank you for your fast reply! Really appreciate your work!!

Staubgeborener, 2024-12-13:
This is something that `--fix` does in the background, among other things. The parameter `--fix` deletes the config folder and thus the .git folder inside.

Most problems of this kind are caused by a misconfiguration of git, which is why the parameter was introduced (and is even mentioned in the issue template).