Staubgeborener/Klipper-Backup #142: cannot access backed config files

YoungHillionaire, 2025-03-12:
### Code of Conduct

- [x] I promise that i have used markdown syntax to make the code more readable which helps to fix the problem and that I have tested the --fix command mentioned above without success.

### What happened

I follow the steps in the docs
i even tried multiple times using the -fix and -f command
and used fine grain token and classic token
i cannot access my backed config on github

[image]

### What did you expect to happen

backup 

[image]

### How to reproduce

--

### Debug

```shell
● Checking for installed dependencies Done!

Klipper-Backup is up to date

------------DEBUG:------------
Command: /home/hussain/klipper-backup/script.sh --debug
------------------------------

------------DEBUG:------------
github_token=****************
github_username=YoungHillionaire
github_repository=Voron2.4
branch_name="main"
commit_username="hussain"
commit_email="7saindhaif@gmail.com"

# All information regarding .env can be found here:
# https://klipperbackup.xyz/configuration/

# Backup paths
# Note: script.sh starts its search in $HOME which is /home/{username}/
# The array accepts folders or files like the following example
#
#  backupPaths=( \
#  "printer_data/config/*" \
#  "printer_data/config/printer.cfg" \
#  )
#
# Using the above example the script will search for `/home/{username}/printer_data/config/*` and `/home/{username}/printer_data/config/printer.cfg`
# When backing up a folder you should always have `/*` at the end of the path so that files insde the folder are properly searched

backupPaths=( \
"printer_data/config/*" \
)

# Array of strings in .gitignore pattern git format https://git-scm.com/docs/gitignore#_pattern_format for files that should not be uploaded to the remote repo
# New additions must be enclosed in double quotes and should follow the pattern format as noted in the above link
exclude=( \
"*.swp" \
"*.tmp" \
"printer-[0-9]*_[0-9]*.cfg" \
"*.bak" \
"*.bkp" \
"*.csv" \
"*.zip" \
------------------------------

------------DEBUG:------------
The GitHub repo YoungHillionaire/Voron2.4 exists (public)
------------------------------

------------DEBUG:------------
$HOME: /home/hussain
------------------------------

------------DEBUG:------------
$backup_path: /home/hussain/config_backup

Content of $backup_path:
total 16
drwxr-xr-x  3 hussain hussain 4096 Mar 13 00:01 .
drwxr-xr-x 27 hussain hussain 4096 Mar 12 23:53 ..
drwxr-xr-x  8 hussain hussain 4096 Mar 13 00:01 .git
-rw-r--r--  1 hussain hussain  181 Mar 12 23:53 README.md
------------------------------

------------DEBUG:------------
$backup_path/.git/config:

[init]
    defaultBranch = main
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[user]
        name = hussain
        email = 7saindhaif@gmail.com
[remote "origin"]
        url = https://********@github.com/YoungHillionaire/Voron2.4.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[branch "main"]
        remote = origin
        merge = refs/heads/main
------------------------------

hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
From https://github.com/YoungHillionaire/Voron2.4
 * branch            main       -> FETCH_HEAD
Already up to date.
Skipping symbolic link: printer_data/config/moonraker_obico_macros.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg

------------DEBUG:------------
Content of $backup_path after rsync:
total 20
drwxr-xr-x  4 hussain hussain 4096 Mar 13 00:04 .
drwxr-xr-x 27 hussain hussain 4096 Mar 12 23:53 ..
drwxr-xr-x  8 hussain hussain 4096 Mar 13 00:04 .git
drwxr-xr-x  3 hussain hussain 4096 Mar 13 00:04 printer_data
-rw-r--r--  1 hussain hussain  181 Mar 12 23:53 README.md
------------------------------

warning: adding embedded git repository: printer_data/config
hint: You've added another git repository inside your current repository.
hint: Clones of the outer repository will not contain the contents of
hint: the embedded repository and will not know how to obtain it.
hint: If you meant to add a submodule, use:
hint:
hint:   git submodule add <url> printer_data/config
hint:
hint: If you added this path by mistake, you can remove it from the
hint: index with:
hint:
hint:   git rm --cached printer_data/config
hint:
hint: See "git help submodule" for more information.
error: object file .git/objects/09/5ddc9a41d4610564518b4126d02402fc296198 is empty
error: object file .git/objects/09/5ddc9a41d4610564518b4126d02402fc296198 is empty
error: object file .git/objects/09/5ddc9a41d4610564518b4126d02402fc296198 is empty
fatal: loose object 095ddc9a41d4610564518b4126d02402fc296198 (stored in .git/objects/09/5ddc9a41d4610564518b4126d02402fc296198) is corrupt
fatal: 'git status --porcelain=2' failed in submodule printer_data/config
[main d764113] New backup from 13/03/25 - 00:04:33 - No new changes pushed
Enumerating objects: 1, done.
Counting objects: 100% (1/1), done.
Writing objects: 100% (1/1), 219 bytes | 219.00 KiB/s, done.
Total 1 (delta 0), reused 0 (delta 0), pack-reused 0
To https://github.com/YoungHillionaire/Voron2.4.git
   96276e5..d764113  main -> main
Branch 'main' set up to track remote branch 'main' from 'origin'.
```

Tylerjet, 2025-03-13:
It seems that your `printer_data/config` folder is a git repo. what is output when you go to the config folder and do ```git log```

YoungHillionaire, 2025-03-13:
hussain@voron:~/printer_data/config $ git log
error: object file .git/objects/09/5ddc9a41d4610564518b4126d02402fc296198 is empty
error: object file .git/objects/09/5ddc9a41d4610564518b4126d02402fc296198 is empty
error: object file .git/objects/09/5ddc9a41d4610564518b4126d02402fc296198 is empty
fatal: loose object 095ddc9a41d4610564518b4126d02402fc296198 (stored in .git/objects/09/5ddc9a41d4610564518b4126d02402fc296198) is corrupt

YoungHillionaire, 2025-03-13:
origin  https://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@github.com/YoungHillionaire/Voron.git/ (fetch)
origin  https://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@github.com/YoungHillionaire/Voron.git/ (push)

YoungHillionaire, 2025-03-13:
I asked chatgpt and it fixed my issue.

If your Klipper printer’s configuration folder is a Git repository and you want to remove the Git tracking, follow these steps:

Option 1: Completely Remove Git Tracking
SSH into your printer
If you're using a Raspberry Pi (e.g., running [Fluidd](https://github.com/fluidd-core/fluidd) or [Mainsail](https://github.com/mainsail-crew/mainsail)), connect via SSH:

ssh pi@your_printer_ip
(Replace pi with your username and your_printer_ip with your printer's IP address.)

Navigate to the Klipper configuration directory
Typically, this is in ~/printer_data/config:

cd ~/printer_data/config

Remove the .git folder
This will delete Git tracking but leave all files intact:

rm -rf .git

Verify Git is removed
Run:

git status
If you see fatal: not a git repository, then Git tracking is successfully removed.

Tylerjet, 2025-03-14:
Pefect yeah i would have linked you to https://github.com/Staubgeborener/Klipper-Backup/issues/121#issuecomment-2356862413 where something similar happened but chatGPT also works.

closed 2025-03-14 by Tylerjet