Staubgeborener/Klipper-Backup #146: Backup not working

rmilyard, 2025-03-24:
### Code of Conduct

- [x] I promise that i have used markdown syntax to make the code more readable which helps to fix the problem and that I have tested the --fix command mentioned above without success.

### What happened

I followed the instructions from Chris' video.  I installed on my Voron doesn't seem to backup anything.  I also did it on  my Ratrig and there is works.  Not sure what might be going on or how to troubleshoot this,

### What did you expect to happen

Should backup

### How to reproduce

NA

### Debug

```shell
pi@DarkLDOpi:~ $ ~/klipper-backup/script.sh --debug
● Checking for installed dependencies Done!

Klipper-Backup is up to date

------------DEBUG:------------
Command: /home/pi/klipper-backup/script.sh --debug
------------------------------

------------DEBUG:------------
github_token=****************
github_username=rmilyard
github_repository=DarkLDO
branch_name="main"
commit_username="pi"
commit_email="rmilyard@hotmail.com"

# All information regarding .env can be found here:
# https://klipperbackup.xyz/configuration/

# Backup paths
# Note: script.sh starts its search in $HOME which is /home/{username}/
# The array accepts folders or files like the following example
#
#  backupPaths=( \
#  "printer_data/config/*" \
#  "printer_data/config/printer.cfg" \
#  )
#
# Using the above example the script will search for `/home/{username}/printer_data/config/*` and `/home/{username}/printer_data/config/printer.cfg`
# When backing up a folder you should always have `/*` at the end of the path so that files insde the folder are properly searched

backupPaths=( \
"printer_data/config/*" \
)

# Array of strings in .gitignore pattern git format https://git-scm.com/docs/gitignore#_pattern_format for files that should not be uploaded to the remote repo
# New additions must be enclosed in double quotes and should follow the pattern format as noted in the above link
exclude=( \
"*.swp" \
"*.tmp" \
"printer-[0-9]*_[0-9]*.cfg" \
"*.bak" \
"*.bkp" \
"*.csv" \
"*.zip" \
------------------------------

------------DEBUG:------------
The GitHub repo rmilyard/DarkLDO exists (public)
------------------------------

------------DEBUG:------------
$HOME: /home/pi
------------------------------

------------DEBUG:------------
$backup_path: /home/pi/config_backup

Content of $backup_path:
total 16
drwxr-xr-x  3 pi pi 4096 Mar 24 12:54 .
drwxr-xr-x 33 pi pi 4096 Mar 24 12:51 ..
drwxr-xr-x  8 pi pi 4096 Mar 24 12:54 .git
-rw-r--r--  1 pi pi  181 Mar 12 13:39 README.md
------------------------------

------------DEBUG:------------
$backup_path/.git/config:

[init]
    defaultBranch = main
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[user]
        name = pi
        email = rmilyard@hotmail.com
[remote "origin"]
        url = https://********@github.com/rmilyard/DarkLDO.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[branch "main"]
        remote = origin
        merge = refs/heads/main
------------------------------

Skipping symbolic link: printer_data/config/KAMP
Skipping symbolic link: printer_data/config/K-ShakeTune
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg

------------DEBUG:------------
Content of $backup_path after rsync:
total 20
drwxr-xr-x  4 pi pi 4096 Mar 24 13:08 .
drwxr-xr-x 33 pi pi 4096 Mar 24 12:51 ..
drwxr-xr-x  8 pi pi 4096 Mar 24 13:08 .git
drwxr-xr-x  3 pi pi 4096 Mar 24 13:08 printer_data
-rw-r--r--  1 pi pi  181 Mar 12 13:39 README.md
------------------------------

fatal: Unable to create '/home/pi/config_backup/.git/index.lock': File exists.

Another git process seems to be running in this repository, e.g.
an editor opened by 'git commit'. Please make sure all processes
are terminated then try again. If it still fails, a git process
may have crashed in this repository earlier:
remove the file manually to continue.
error: object file .git/objects/e1/095e2d5b947b6fe547224c2cd8c6e5ff9ca20b is empty
error: object file .git/objects/e1/095e2d5b947b6fe547224c2cd8c6e5ff9ca20b is empty
fatal: loose object e1095e2d5b947b6fe547224c2cd8c6e5ff9ca20b (stored in .git/objects/e1/095e2d5b947b6fe547224c2cd8c6e5ff9ca20b) is corrupt
error: object file .git/objects/e1/095e2d5b947b6fe547224c2cd8c6e5ff9ca20b is empty
error: object file .git/objects/e1/095e2d5b947b6fe547224c2cd8c6e5ff9ca20b is empty
error: object file .git/objects/e1/095e2d5b947b6fe547224c2cd8c6e5ff9ca20b is empty
fatal: loose object e1095e2d5b947b6fe547224c2cd8c6e5ff9ca20b (stored in .git/objects/e1/095e2d5b947b6fe547224c2cd8c6e5ff9ca20b) is corrupt
fatal: the remote end hung up unexpectedly
send-pack: unexpected disconnect while reading sideband packet
error: failed to push some refs to 'https://github.com/rmilyard/DarkLDO.git'
fatal: the remote end hung up unexpectedly
```

rmilyard, 2025-03-24:
I am not sure what issue might be here.  Can someone help me out with this?

Tylerjet, 2025-03-25:
```fatal: Unable to create '/home/pi/config_backup/.git/index.lock': File exists.

Another git process seems to be running in this repository, e.g.
an editor opened by 'git commit'. Please make sure all processes
are terminated then try again. If it still fails, a git process
may have crashed in this repository earlier:
```

The issue looks to be in the config_backup folder were you trying to make changes or configure something in there? As the error states that another git process is running in that folder.

rmilyard, 2025-03-25:
When runs now I only get config folder but empty.

Tylerjet, 2025-03-25:
I would probably say try and delete the config_backup folder and then try running script.sh again

Tylerjet, 2025-04-01:
@rmilyard was that able to resolve it?

Achduka, 2025-04-07:
Had the same issue and deleting the config_backup directory solved it.

rmilyard, 2025-04-07:
I deleted the same and still not working

Tylerjet, 2025-04-08:
> I deleted the same and still not working

Same error or a different error?

Tylerjet, 2025-04-09:
Can you post another log of the debug run?

Staubgeborener, 2025-04-09:
like you did before `~/klipper-backup/script.sh --debug`

rmilyard, 2025-04-09:
'
● Checking for installed dependencies Done!

Klipper-Backup is up to date

------------DEBUG:------------
Command: /home/pi/klipper-backup/script.sh --debug
------------------------------

------------DEBUG:------------
github_token=****************
github_username=rmilyard
github_repository=DarkLDO
branch_name="main"
commit_username="pi"
commit_email="rmilyard@hotmail.com"

# All information regarding .env can be found here:
# https://klipperbackup.xyz/configuration/

# Backup paths
# Note: script.sh starts its search in $HOME which is /home/{username}/
# The array accepts folders or files like the following example
#
#  backupPaths=( \
#  "printer_data/config/*" \
#  "printer_data/config/printer.cfg" \
#  )
#
# Using the above example the script will search for `/home/{username}/printer_data/config/*` and `/home/{username}/printer_data/config/printer.cfg`
# When backing up a folder you should always have `/*` at the end of the path so that files insde the folder are properly searched

backupPaths=( \
"printer_data/config/*" \
)

# Array of strings in .gitignore pattern git format https://git-scm.com/docs/gitignore#_pattern_format for files that should not be uploaded to the remote repo
# New additions must be enclosed in double quotes and should follow the pattern format as noted in the above link
exclude=( \
"*.swp" \
"*.tmp" \
"printer-[0-9]*_[0-9]*.cfg" \
"*.bak" \
"*.bkp" \
"*.csv" \
"*.zip" \
------------------------------

------------DEBUG:------------
The GitHub repo rmilyard/DarkLDO exists (public)
------------------------------

------------DEBUG:------------
$HOME: /home/pi
------------------------------

------------DEBUG:------------
$backup_path: /home/pi/config_backup

Content of $backup_path:
total 16
drwxr-xr-x  3 pi pi 4096 Apr  9 08:00 .
drwxr-xr-x 33 pi pi 4096 Apr  9 07:35 ..
drwxr-xr-x  8 pi pi 4096 Apr  9 08:00 .git
-rw-r--r--  1 pi pi  181 Apr  7 16:01 README.md
------------------------------

------------DEBUG:------------
$backup_path/.git/config:

[init]
    defaultBranch = main
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[user]
        name = pi
        email = rmilyard@hotmail.com
[remote "origin"]
        url = https://********@github.com/rmilyard/DarkLDO.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[branch "main"]
        remote = origin
        merge = refs/heads/main
------------------------------

hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
From https://github.com/rmilyard/DarkLDO
 * branch            main       -> FETCH_HEAD
Already up to date.
Skipping symbolic link: printer_data/config/KAMP
Skipping symbolic link: printer_data/config/K-ShakeTune
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg

------------DEBUG:------------
Content of $backup_path after rsync:
total 20
drwxr-xr-x  4 pi pi 4096 Apr  9 08:09 .
drwxr-xr-x 33 pi pi 4096 Apr  9 07:35 ..
drwxr-xr-x  8 pi pi 4096 Apr  9 08:09 .git
drwxr-xr-x  3 pi pi 4096 Apr  9 08:09 printer_data
-rw-r--r--  1 pi pi  181 Apr  7 16:01 README.md
------------------------------

warning: adding embedded git repository: printer_data/config
hint: You've added another git repository inside your current repository.
hint: Clones of the outer repository will not contain the contents of
hint: the embedded repository and will not know how to obtain it.
hint: If you meant to add a submodule, use:
hint:
hint:   git submodule add <url> printer_data/config
hint:
hint: If you added this path by mistake, you can remove it from the
hint: index with:
hint:
hint:   git rm --cached printer_data/config
hint:
hint: See "git help submodule" for more information.
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
        modified:   printer_data/config (modified content)

no changes added to commit (use "git add" and/or "git commit -a")
[main 2f5fc31] New backup from 09/04/25 - 08:09:15 - No new changes pushed
Enumerating objects: 1, done.
Counting objects: 100% (1/1), done.
Writing objects: 100% (1/1), 216 bytes | 216.00 KiB/s, done.
Total 1 (delta 0), reused 0 (delta 0), pack-reused 0
To https://github.com/rmilyard/DarkLDO.git
   bf350c7..2f5fc31  main -> main
Branch 'main' set up to track remote branch 'main' from 'origin'.

'

Staubgeborener, 2025-04-09:
Thats interesting. Basically everything works but looking deeper in your [commit history](https://github.com/rmilyard/DarkLDO/commits/main/printer_data) i clearly see that every backup works perfectly. But then comes a cron backupo ([like](https://github.com/rmilyard/DarkLDO/commit/bf350c73d58d484706a12564458d9ef34a6fb97c) this or [this](https://github.com/rmilyard/DarkLDO/commit/0c90ef1f4c1ac5c6bdd67c624d1654fd5e6fb2a9)) that nukes everything.

If I look at the timestamps from Apr 8, 2025 the cron is run every 4 hours so I guess it will be [this one](https://klipperbackup.xyz/automation/#cron)?

My recommendation to fix this:
1. Enter `crontab -e` and give us the whole output (or just a screenshot), so we can investigate it later.
2. **AFTER THAT** delete the line which looks like this `0 */4 * * * $HOME/klipper-backup/script.sh` and save it (STRG +X -> Y[es])
3. To run backups automatically in the future, simply use the systemd command to run sscript on every startup:
[https://klipperbackup.xyz/automation/#backup-on-boot](https://klipperbackup.xyz/automation/#backup-on-boot)

rmilyard, 2025-04-09:
[image]

rmilyard, 2025-04-09:
Just did above and still same.  Nothing but config folder that is empty is in backup.

rmilyard, 2025-04-09:
Still empty.  

~/klipper-backup/script.sh
● Checking for installed dependencies Done!

● Delete config_backup folder Done!

● Checking for installed dependencies Done!

Klipper-Backup is up to date

Initialized empty Git repository in /home/pi/config_backup/.git/
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
remote: Enumerating objects: 70, done.
remote: Counting objects: 100% (70/70), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 70 (delta 34), reused 36 (delta 0), pack-reused 0 (from 0)
Unpacking objects: 100% (70/70), 193.66 KiB | 1.29 MiB/s, done.
From https://github.com/rmilyard/DarkLDO
 * branch            main       -> FETCH_HEAD
 * [new branch]      main       -> origin/main
Skipping symbolic link: printer_data/config/KAMP
Skipping symbolic link: printer_data/config/K-ShakeTune
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg
warning: adding embedded git repository: printer_data/config
hint: You've added another git repository inside your current repository.
hint: Clones of the outer repository will not contain the contents of
hint: the embedded repository and will not know how to obtain it.
hint: If you meant to add a submodule, use:
hint:
hint:   git submodule add <url> printer_data/config
hint:
hint: If you added this path by mistake, you can remove it from the
hint: index with:
hint:
hint:   git rm --cached printer_data/config
hint:
hint: See "git help submodule" for more information.
On branch main
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
        modified:   printer_data/config (modified content)

no changes added to commit (use "git add" and/or "git commit -a")
From https://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@github.com/rmilyard/DarkLDO.git
Branch 'main' set up to track remote branch 'main' from 'origin'.
Everything up-to-date
pi@DarkLDOpi:~ $

Staubgeborener, 2025-04-09:
I would also recommend it, your entire commit history is a mess. Current commits are displayed under older commits, there is no chronological order. Very strange. I don't know any users who has the same issue as you.

**Remove everything:**
```shell
sudo rm -r ~/klipper-backup
sudo rm -r ~/config_backup

sudo systemctl stop klipper-backup-on-boot.service
sudo systemctl disable klipper-backup-on-boot.service
rm /etc/systemd/system/klipper-backup-on-boot.service

sudo systemctl stop klipper-backup-filewatch.service
sudo systemctl disable klipper-backup-filewatch.service
rm /etc/systemd/system/klipper-backup-on-boot.service
```
Please again, check `crontab -e` that everything regarding Klipper-Backup is removed.

**Installation:**
Now we start from scratch. **Don't install** backup on boot, filewatch or the 4 hours cronjob. We can do this later as soon as everything is up and running. Please also delete your GitHub [repository](https://github.com/rmilyard/DarkLDO/settings) (scroll all the way down to the "Delete this repository" button) just in case. Your new repository can also be named `DarkLDO`. Create a new [token](https://klipperbackup.xyz/installation/#create-github-token) for the new repository.

```shell
curl -fsSL get.klipperbackup.xyz | bash
~/klipper-backup/install.sh
```

Again: Say no to backup on boot, filewatch or the 4 hours cronjob. So far, your old env looked good, everything is correct.

```
github_username=rmilyard
github_repository=DarkLDO
branch_name="main"
commit_username="pi"
commit_email="rmilyard@hotmail.com"
```

Staubgeborener, 2025-04-09:
Could you please show the following
```
ls -la ~
ls -la ~/printer_data
ls -la ~/printer_data/config
```

rmilyard, 2025-04-09:
pi@DarkLDOpi:~ $ ls -la ~
ls -la ~/printer_data
ls -la ~/printer_data/config
total 236
drwxr-xr-x 33 pi   pi    4096 Apr  9 09:27 .
drwxr-xr-x  3 root root  4096 May  2  2023 ..
drwxr-xr-x 12 pi   pi    4096 Mar 24 06:49 AFC-Klipper-Add-On
-rw-------  1 pi   pi   41762 Apr  9 09:27 .bash_history
-rw-r--r--  1 pi   pi     220 May  2  2023 .bash_logout
-rw-r--r--  1 pi   pi    3523 May  2  2023 .bashrc
drwxr-xr-x  5 pi   pi    4096 May 20  2024 .cache
drwxr-xr-x  3 pi   pi    4096 May 20  2024 .config
drwxr-xr-x  3 pi   pi    4096 Apr  9 09:26 config_backup
drwxr-xr-x 10 pi   pi    4096 Feb 28 06:44 crowsnest
-rw-r--r--  1 pi   pi      49 Feb 13  2024 .gitconfig
-rw-r--r--  1 pi   pi   12631 Feb  7 11:00 install-moonraker.sh
-rw-r--r--  1 pi   pi   12631 Feb  7 11:00 install-moonraker.sh.1
drwxr-xr-x  7 pi   pi    4096 Mar 26  2024 katapult
drwxr-xr-x  4 pi   pi    4096 Mar 25  2024 katapult-env
drwxr-xr-x  8 pi   pi    4096 Apr  2 06:54 kiauh
drwxr-xr-x  3 pi   pi    4096 Jan 12  2024 kiauh-backups
-rw-r--r--  1 pi   pi     581 Apr  9 09:11 .kiauh.ini
drwxr-xr-x  8 pi   pi    4096 Feb 24 06:40 klippain_shaketune
drwxr-xr-x  6 pi   pi    4096 May 20  2024 klippain_shaketune-env
drwxr-xr-x 12 pi   pi    4096 Apr  9 07:36 klipper
drwxr-xr-x  6 pi   pi    4096 Mar 12 14:18 Klipper-Adaptive-Meshing-Purging
drwxr-xr-x  6 pi   pi    4096 Apr  9 09:24 klipper-backup
drwxr-xr-x  2 pi   pi    4096 Feb 28 06:48 klipper-kconfigs
drwxr-xr-x  9 pi   pi    4096 Oct  2  2024 klipper-led_effect
drwxr-xr-x 10 pi   pi    4096 Feb 25 07:45 KlipperScreen
drwxr-xr-x  5 pi   pi    4096 Jan 12  2024 .KlipperScreen-env
drwxr-xr-x  3 pi   pi    4096 Jan 19 12:55 klipper_tmc_autotune
drwxr-xr-x  5 pi   pi    4096 Feb 25 07:45 klippy-env
drwxr-xr-x  6 pi   pi    4096 Feb 25 07:44 klippy-env2.7
drwxr-xr-x  5 pi   pi    4096 Feb  7 11:05 .local
drwxr-xr-x  6 pi   pi    4096 Dec 26 08:06 mainsail
drwxr-xr-x  3 pi   pi    4096 Nov 25 07:52 mainsail-config
drwxr-xr-x  8 pi   pi    4096 Apr  4 08:00 moonraker
drwxr-xr-x  4 pi   pi    4096 May 25  2023 moonraker-env
drwxr-xr-x  8 pi   pi    4096 May 25  2023 moonraker-timelapse
drwxr-xr-x  5 pi   pi    4096 Jan 12  2024 pgcode
drwxr-xr-x 11 pi   pi    4096 Feb  9  2024 printer_data
-rw-r--r--  1 pi   pi     807 May  2  2023 .profile
-rw-r--r--  1 pi   pi      66 Apr  9 08:32 .selected_editor
drwxr-xr-x  8 pi   pi    4096 Jan 19  2024 sonar
drwxr-xr-x  2 pi   pi    4096 Jan 25  2024 timelapse
-rw-r--r--  1 pi   pi     215 Mar 10 13:09 .wget-hsts
total 52
drwxr-xr-x 11 pi pi 4096 Feb  9  2024 .
drwxr-xr-x 33 pi pi 4096 Apr  9 09:27 ..
drwxr-xr-x  3 pi pi 4096 May 13  2023 backup
drwxr-xr-x  2 pi pi 4096 May 13  2023 certs
drwxr-xr-x  2 pi pi 4096 Apr  9 09:24 comms
drwxr-xr-x  6 pi pi 4096 Apr  2 09:09 config
drwxr-xr-x  2 pi pi 4096 Apr  9 09:24 database
drwxr-xr-x  3 pi pi 4096 Apr  2 13:51 gcodes
drwxr-xr-x  2 pi pi 4096 Apr  9 07:31 logs
drwxr-xr-x  2 pi pi 4096 Feb  1  2024 misc
-rw-r--r--  1 pi pi  131 May 13  2023 moonraker.asvc
-rw-r--r--  1 pi pi   32 Jan 12  2024 .moonraker.uuid
drwxr-xr-x  2 pi pi 4096 Jan 12  2024 systemd
total 144
drwxr-xr-x  6 pi pi  4096 Apr  2 09:09 .
drwxr-xr-x 11 pi pi  4096 Feb  9  2024 ..
drwxr-xr-x  4 pi pi  4096 Mar 25 15:50 AFC
-rw-r--r--  1 pi pi 12439 Feb 13  2024 Clean_Nozzle.cfg
-rw-r--r--  1 pi pi  3167 May  9  2024 crowsnest.conf
drwxr-xr-x  8 pi pi  4096 Feb 10 11:56 .git
-rw-r--r--  1 pi pi    80 Feb 13  2024 .gitignore
lrwxrwxrwx  1 pi pi    55 Feb  9  2024 KAMP -> /home/pi/Klipper-Adaptive-Meshing-Purging/Configuration
-rw-r--r--  1 pi pi  3006 Mar 12 14:19 KAMP_Settings.cfg
-rw-r--r--  1 pi pi   209 Apr  9 08:17 KlipperScreen.conf
-rw-r--r--  1 pi pi  1550 Feb  8  2024 knomi.cfg
lrwxrwxrwx  1 pi pi    36 May 20  2024 K-ShakeTune -> ../../klippain_shaketune/K-ShakeTune
drwxr-xr-x  5 pi pi  4096 May 20  2024 K-ShakeTune_results
-rw-r--r--  1 pi pi  1065 Feb 13  2024 LICENSE
lrwxrwxrwx  1 pi pi    37 May 25  2023 mainsail.cfg -> /home/pi/mainsail-config/mainsail.cfg
-rw-r--r--  1 pi pi  2748 Apr  9 09:24 moonraker.conf
-rw-r--r--  1 pi pi  2595 Apr  9 09:24 .moonraker.conf.bkp
-rw-r--r--  1 pi pi  9487 Feb  7  2024 nitehawk-sb.cfg
-rw-r--r--  1 pi pi 26330 Apr  2 09:09 printer.cfg
-rw-r--r--  1 pi pi  8629 Mar 13  2024 Sensorless-Homing.cfg
-rw-r--r--  1 pi pi   770 May 25  2023 sonar.conf
-rw-r--r--  1 pi pi  9862 Feb  7  2024 stealthburner_led_effects_barf.cfg
-rw-r--r--  1 pi pi   918 Feb  7  2024 tacho_macros.cfg
drwxr-xr-x  5 pi pi  4096 Jan 19  2024 .theme
lrwxrwxrwx  1 pi pi    56 May 25  2023 timelapse.cfg -> /home/pi/moonraker-timelapse/klipper_macro/timelapse.cfg
pi@DarkLDOpi:~ $

Staubgeborener, 2025-04-09:
Please one command after another, not all at the same time.

Staubgeborener, 2025-04-09:
Please one command after another, not all at the same time. Right now we only have the `ls -la ~/printer_data/config` output.

datapaganism, 2025-04-22:
> What I can already tell you: This is 100% not a Klipper-Backup problem, but actually specific to your setup. I'll explain why later. But let's try to solve it anyway.
> 
> I think that `ls -la ~/printer_data/config` shows the issue. There is a hidden .git folder and also a hidden .gitignore file.
> 
> ```
> pi@DarkLDOpi:~ $ ls -la ~/printer_data/config
> [...]
> total 144
> [...]
> drwxr-xr-x  8 pi pi  4096 Feb 10 11:56 .git
> -rw-r--r--  1 pi pi    80 Feb 13  2024 .gitignore
> ```
> 
> Why? I have no idea.
> 
> First, we create a local backup, just to be on the safe side: `cp -r ~/printer_data/config ~/klipper-config-backup`. Now we try something:
> 
> cd ~/printer_data
> git rm --cached config
> rm -rf ./config/.git
> ~/klipper-backup/script.sh

Thank you this solved my issue

Staubgeborener, 2025-04-22:
Memo to myself: add a debug check to see if `.git` folders exist in printer_data directory

closed 2025-04-22 by Staubgeborener

Staubgeborener, 2025-05-25:
@masteryodha We already thought about excluding the .git folder. Over a year ago we tried this and [reverse these changes immediately](https://github.com/Staubgeborener/Klipper-Backup/commit/22b1d5b25fa1344b80436684a835d2b848d66fe0). Good news, I added a comment in the description why we changed it back:

> undo --exclude=".git" as this would lead to future restoring problems

I also talked with @Tylerjet a while ago about adding this. From my side, we could do that, but @Tylerjet is more into the restore thing and can explain if this will break something later (or not).

Tylerjet, 2025-05-25:
But realistically any config repos like that should follow the same structure as other addons where the repos are placed outside of the config folder, and are symlinked to the location instead, then you create your own overrides/modification within the config folder which is backed up as normal.