Staubgeborener/Klipper-Backup #148: The other day I messed up a pull request, but this is what I was working on

Bradford1040, 2025-04-25:
### Is your feature request related to a problem? Please describe.

[klipper-backup](https://github.com/Bradford1040/Klipper-Backup/tree/devel-v3.0) for KIAUH installed multiple printers with custom names, not printer1_data, printer2_data, but if you choose custom names. like punisher_data, able_data, kane_data, and so forth

[image of ssh](https://photos.app.goo.gl/m4jADtLPAvRATC8A9)

Not only is the folder different but klipper.service & moonraker.service gets their own custom name for each printer, I have 12 printers with only 8 at the moment are active, but like above each has its own service, klipper-punisher.service, klipper-able.service, same with moonraker, moonraker-punisher.service, moonraker-able.service.

I first set up symbolic links from the main install ~/klipper-backup to each folder and edited the .env file for each and set up repo's for each on GitHub, but filewatch and on-boot only set up one printer, and if I remember correctly I had to edit that as well.

I don't use a R-pi I use a server or a PC as there is no way a R-pi could handle the amount of USB connection I have, between printers, cameras, accelerometers, BTT-EDDY's, and so forth. Plus the little SOC on the R-pi would most likely explode lol! 

I am using RHINO-LINUX Distro which is a Debian/Ubuntu rolling release, which has its own issues trust me. Python dependencies were a nightmare in klippain-shaketune, which started me on my repo editing, before that. Truly I barely have done anything for years, and even then it was minecraft stuff!

I was going to install each printer as its own cron, but that is down the road if at all, I'm sure to have a few glitches that maybe you'll see, as I am still in the kiddy pool, please take that into consideration when looking over it, oh also the reason I have the token bypassed in `script.sh` with `--no-gpg-sign` as I have verified commit signing on my GitHub and I think that was my only option at the moment. 

Thank you for all your hard work, I hope that maybe this can maybe give you a few ideas, as well as any you feel like throwing my way lol! I have been working on it for 2 weeks steady now, before it was just editing yours but I got to thinking and reading, and this is the result 

### Describe the solution you'd like

Solution is above I saw this at first as optional, sorry for another mess up lol

### Describe alternatives you've considered

I included above as well, maybe I should of edited but it is almost 4am 

### Additional information

I am on many discord servers, mainly I mostly hang on ZeroG mercury 1 server, most my info is on my main GH page, if you wish to chat

Staubgeborener, 2025-04-25:
Let's break this down:

We have created an [article in our documentation](https://klipperbackup.xyz/faq/#klipper-backup-with-multiple-instances) about using Klipper-Backup with multiple printers.
Normally, if you set up several printers with KIAUH, each printer should have its own directory in the home directory. The name of the directory is the name of the printer. So in your example, something like `/home/octo/punisher_data`, `/home/octo/able_data`, `/home/octo/kane_data`. All printer-specific data for the respective printer is in the respective folder. Which means you can adjkust your .env like this:
```ini
backupPaths=( \
"kane_data/config/*" \
"punisher_data/config/*" \
"able_data/config/*"
)
```

I hope that has answered the first half for you. About the second half: cron and services.
> Not only is the folder different but klipper.service & moonraker.service gets their own custom name for each printer, I have 12 printers with only 8 at the moment are active, but like above each has [...]

It doesn't matter, Klipper-Backup is not interested in this data. These are not touched, nothing is read, nothing is saved. So you can ignore it.

> I was going to install each printer as its own cron

Not necessary. The cronjob or the service files only do one thing: execute `script.sh`. And `script.sh` in turn looks in your .env, where all paths are defined for each individual printer. If you were to write a cronjob for each printer, for example, you would have a bunch of redundant files that all do the same thing: execute `script.sh`.

I may not have understood your request 100% correctly, but if I did, the solution would be to look in your home directory to see how your printers are named and adjust your .env file accordingly. Leave cron and services alone and everything will work.

Bradford1040, 2025-04-26:
# I wish I seen your change a while ago this might of had me go in a different direction lol

```shell
backupPaths=( \
"kane_data/config/*" \
"punisher_data/config/*" \
"able_data/config/*"
)
```

But the way I was doing it before I went through and did all this was symbolic links, see I did not know about adding to the `.env`  plus I wanted a different repo for each printer, not all under the same repo, mostly because a few of the printers I have going are now owned by another person's and I upload to their repo's respectively, as I have allowed them access to make changes to their UI and settings.

again I have;

*  Dell Poweredge R730xd
*  Rhino Linux 
*  Webmin & Usermin 
*  Operating system | Debian Linux 13.0/Ubuntu 25.04 rolling release
*  Linux 6.14.2-061402-generic on x86_64
*  Ubuntu AUR repository (mainly)

But I also wasn't really requesting you to make any changes, your application is great! (I do wish I had tried out KIAUH v6 again) I had forgotten that I opted out when I was using a different distro as there were issues with KIAUH v6 and he seemed to not care about the errors I was receiving, it was one of those "it is not my fault, it's theirs" & the other app dev said the same thing, so rather than get in the middle I just opted out of V6 last year, and forgot about it as I only really use it for original installation and then I do everything myself, as I run mostly forked repositories because of custom printers that each require their own little tweaks at some level

## I basically said I would share what I was working on the other day when I messed up my spellcheck pull request, I had fallen asleep and sent the request after I got up thinking it was ready and I included my .vscode by accident. I am just sharing my work
as a courtesy, it might be over engineered, but I am just learning (at 54 yrs old)

Staubgeborener, 2025-04-27:
> I wanted a different repo for each printer, not all under the same repo

Currently, not really natively possible, the script only accepts a repository in the .env as input, no more.
The only possibility I can think of is to install Klipper-Backup several times. Each printer gets its own Klipper-Backup and therefore its own .env and thus its own repository URL. But this will not work with KIAUH. You have to do something like

`git clone https://github.com/Staubgeborener/Klipper-Backup Klipper-Backup_kane_data`
`git clone https://github.com/Staubgeborener/Klipper-Backup Klipper-Backup_punisher_data`
`git clone https://github.com/Staubgeborener/Klipper-Backup Klipper-Backup_able_data`

This will create 3 folders in your home directory called Klipper-Backup_kane_data, Klipper-Backup_punisher_data and Klipper-Backup_able_data. Each will have the same content at first. Navigate into each folder and adjust the .env file, like adding the specific GitHub token and repository URL.

It's a bit tricky and a workaround, but technically possible.

> but I am just learning (at 54 yrs old) 

Nice, keep up the work!

Bradford1040, 2025-04-29:
My edit of your app, allows for it, it does it dynamically. I am still testing it of course but it works pretty good, when you run the `./install.sh` from `$HOME/klipper-backup` it asks you what folder you want to install it in, then clones the repo into that folder, creates the `moonraker.conf` & `shell_command.cfg` if not already present, and creates the `klipper-backup-filewatch-<custom_name>.service` removal of `_data` and does the same for the `on-boot` 

I just wanted to share with you as I said I would, if you have a chance check out `devel-v3.0` branch I think I included that link in the original post.

Bradford1040, 2025-05-03:
I had a few glitches at first but after about 20 test installs I think I got it, one issue I ran into was with the `cron_job` but ending up adding a `next_minute` scanning the previous installed and adding 2 minutes to it, so say the first job `0 */4 * * *` the next install would be `2 */4 * * *` that way the jobs run right after each other. 

Plus I added `git status --untracked-files=all` to it, I was having an odd issue. Took me days to figure it out, and truly I really didn't this just worked is all lol, I also am finding out I am much better at shell than python! I am going to close this, as I am sure you have better stuff to do than chat with me, I plan on lurking on your work lol, so I shall be around. Again Thank you, for a great application, truly great.

closed 2025-05-03 by Bradford1040