Staubgeborener/Klipper-Backup #149: fatal: bad config line 1 in file .git/config

iparsw, 2025-04-26:
it was working fine until I saw it wasn't working for past 3 months
When i try to update it says :

"Klipper-Backup is up to date"
"fatal: bad config line 1 in file .git/config" (2 times)
"Branch: main in .env does not match the currently checked out branch of: ."
"fatal: bad config line 1 in file .git/config" (7 times)
"skipping non-regular file "printer_data/config/CONFIG EXTRA/mainsail.cfg"" (which I don't want to skip)
"Skipping symbolic link: printer_data/config/KAMP" (which i again dont want to skip because I customized it)
"Skipping symbolic link: printer_data/config/print_area_bed_mesh.cfg"
"Skipping symbolic link: printer_data/config/timelapse.cfg"
"Empty source arg specified.
rsync error: syntax or usage error (code 1) at main.c(1508) [sender=3.2.7]" (2 times)
"fatal: bad config line 1 in file .git/config" (6 more times)
my .env

`
github_token=************************************* (its correct)
github_username=iparsw
github_repository=i3_klipper
branch_name="main"
commit_username="klipper_backup"
commit_email="[parsaroshanak@gmail.com](mailto:parsaroshanak@gmail.com)"

backupPaths=(
"printer_data/config/"
"printer_data/config/CONFIG EXTRA/"
"printer_data/config/KAMP/*"
)

exclude=(
".swp"
".tmp"
"printer-[0-9]_[0-9].cfg"
".bak"
".bkp"
".csv"
".zip"
)
`

my .git/config
`[core] repositoryformatversion = 0 filemode = true bare = false logallrefupdates = true 
[remote "origin"] url = https://github.com/Staubgeborener/klipper-backup.git fetch = +refs/heads/*:refs/remotes/origin/* 
[branch "main"] remote = origin merge = refs/heads/main `

my knowledge about git is almost 0 so I'm sorry if the problem is obvious

iparsw, 2025-04-26:
I think i have fixed the issue but im not sure how and what caused it
i ran ```~/klipper-backup/script.sh --fix``` 3 or 4 times 
then i ran ```~/klipper-backup/script.sh```
it updates but it gives these errors and warnings:
```
Checking for installed dependencies Done!

Klipper-Backup is up to date

From https://github.com/iparsw/i3_klipper
 * branch            main       -> FETCH_HEAD
Already up to date.
skipping non-regular file "printer_data/config/CONFIG EXTRA/mainsail.cfg"
Skipping symbolic link: printer_data/config/KAMP
Skipping symbolic link: printer_data/config/print_area_bed_mesh.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg
Empty source arg specified.
rsync error: syntax or usage error (code 1) at main.c(1508) [sender=3.2.7]
Empty source arg specified.
rsync error: syntax or usage error (code 1) at main.c(1508) [sender=3.2.7]
warning: adding embedded git repository: printer_data/config/.theme
hint: You've added another git repository inside your current repository.
hint: Clones of the outer repository will not contain the contents of
hint: the embedded repository and will not know how to obtain it.
hint: If you meant to add a submodule, use:
hint:
hint:   git submodule add <url> printer_data/config/.theme
hint:
hint: If you added this path by mistake, you can remove it from the
hint: index with:
hint:
hint:   git rm --cached printer_data/config/.theme
hint:
hint: See "git help submodule" for more information.
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
[main 7498a9a] New backup from 04/26/2025 - 11:28:09 AM - No new changes pushed
Enumerating objects: 1, done.
Counting objects: 100% (1/1), done.
Writing objects: 100% (1/1), 230 bytes | 230.00 KiB/s, done.
Total 1 (delta 0), reused 0 (delta 0), pack-reused 0
To https://github.com/iparsw/i3_klipper.git
   b93fa45..7498a9a  main -> main
branch 'main' set up to track 'origin/main'.
```

Staubgeborener, 2025-04-27:
Your original mistake happens when you play around with git. For example, changing repositories (e.g. the name and thus the URL), sometimes locally in the .env, sometimes online in GitHub itself, etc. There are several constellations how this can happen but can be broken down to: git problems.

Most users of this script have no idea about git, so we have simply introduced the parameter `--fix`. This simply deletes the `config_backup` folder and thus also the .git folder in it. All the git-relevant stuff is stored in the .git folder. By deleting it, we force a kind of “hard reset” and it works again. [More information here](https://klipperbackup.xyz/faq/#fix-git-errors).

> skipping non-regular file "printer_data/config/CONFIG EXTRA/mainsail.cfg" (which I don't want to skip)
Skipping symbolic link: printer_data/config/KAMP (which i again dont want to skip because I customized it)

You can find out more about symbolic links [here](https://klipperbackup.xyz/faq/#symolic-links). If the file is really important to you, you must include the original file (**NOT THE SYMBOLIC LINK!**) in the .env. The link tells you how to find it and how it works.

If this problem is solved, I would ask you to close the issue via the "Close" button. If there is anything else, just ask.

closed 2025-05-26 by github-actions[bot]