Staubgeborener/Klipper-Backup #155: Minor typo in wiki

jmkaz16, 2025-08-20:
### Code of Conduct

- [x] I promise that i have used markdown syntax to make the code more readable which helps to fix the problem and that I have tested the --fix command mentioned above without success.

### What happened

There is a typo in the FAQ section of the official wiki:
https://klipperbackup.xyz/faq/#symolic-links

The heading currently says **"Symolic links"** instead of **"Symbolic links".**

### What did you expect to happen

The FAQ should display the correct term: **Symbolic links.**

### How to reproduce

1. Go to https://klipperbackup.xyz/faq/#symolic-links
2. Check the section title, it reads **"Symolic links"** instead of **"Symbolic links".**

### Debug

```shell
(Not applicable, as this is a documentation typo.)
```

### Additional information

You might also consider clarifying the expression “The tl;dr version” for non-English speakers. Hope this helps :)

[image]

Tylerjet, 2025-08-21:
Was probably my fault when fixing links a bit ago should be good now

closed 2025-08-21 by Tylerjet

Staubgeborener, 2025-08-21:
Thanks for the hints, it's also possible to open a [PR](https://github.com/Staubgeborener/Klipper-Backup/pulls) if you find something in the FAQ.

> Was probably my fault when fixing links a bit ago should be good now

**>:(**

Staubgeborener, 2025-08-24:
> Where can I find the source files for the wiki? Are they part of this repository, or are they hosted separately? I’d be happy to help out and open a PR next time.

Extra [docs branch](https://github.com/Staubgeborener/Klipper-Backup/tree/docs) in this repository.

jmkaz16, 2025-08-24:
> Extra [docs branch](https://github.com/Staubgeborener/Klipper-Backup/tree/docs) in this repository.

Thanks!